leggedrobotics/rsl_rl #20: time out bootstrapping possible bug.

HyunyoungJung, 2023-12-24:
Hi, thank you for sharing this amazing code.

Recently, I've been looking into the detailed implementation of the code in relation to the paper "Learning to Walk in Minutes Using Massively Parallel Deep Reinforcement Learning."

From my understanding, based on the paper, the reward function is bootstrapped in the event of a time-out. I believe this bootstrapping should apply to the subsequent state, following the formula:
$r_{new} = r + v(s')$,
$s'$ represents the state resulting from the current step.
However, in the current implementation of the code, it appears to be executed as:
$r_{new} = r + v(s)$, 
where $s$ is the state used for the current step.

Could you please clarify if my understanding aligns with the intended design?
I am curious to know whether this implementation choice was deliberate for specific reasons or if it might be an oversight.

Thank you for your time and assistance.

mohakbhardwaj, 2024-11-01:
Hi, I second this issue and have posted a related one:
 https://github.com/leggedrobotics/rsl_rl/issues/43

ClemensSchwarke, 2025-05-15:
Hi @HyunyoungJung, could you point me to the lines you are referring to?

ClemensSchwarke, 2025-05-26:
Closing due to inactivity.

closed 2025-05-26 by ClemensSchwarke

shunlinlu, 2025-11-07:
Anyone solved this problem?