leggedrobotics/rsl_rl #26: mini_batch_generator, same indice in different epoch

swstbecrpgmail, 2024-04-30:
The index generation operation for dataset is placed outside the loop, resulting that the data slices for each epoch remains the same. Is this a problem, or is this a less-than-ideal implementation?

The code is in /rsl_rl/storage/rollout_storage.py
[image]

mahanxin, 2025-02-13:
> The index generation operation for dataset is placed outside the loop, resulting that the data slices for each epoch remains the same. Is this a problem, or is this a less-than-ideal implementation?
> 
> The code is in /rsl_rl/storage/rollout_storage.py [image]

@Mayankm96   What are the benefits of this modification? Can you answer the following? Thank you.

Mayankm96, 2025-02-25:
While resampling minibatch indices for every epoch is the "right" way, we do not do that since it would marginally improve the performance. The number of mini-batches is small while the size of each mini-batch is large. This means that the variance of the transition tuples in each mini-batch would be "similar" and we don't require that much resampling per epoch. The original reason could also be that `randperm` is expensive.

However, if you find that sampling in every epoch leads to more stable training, please feel free to send an MR :)

mahanxin, 2025-02-28:
> While resampling minibatch indices for every epoch is the "right" way, we do not do that since it would marginally improve the performance. The number of mini-batches is small while the size of each mini-batch is large. This means that the variance of the transition tuples in each mini-batch would be "similar" and we don't require that much resampling per epoch. The original reason could also be that `randperm` is expensive.虽然为每个 epoch 重新采样小批量索引是“正确”的方法，但我们不这样做，因为它会略微提高性能。小批量的数量较少，而每个小批量的大小较大。这意味着每个小批量中 transition Tuples 的方差将是 “相似的”，我们不需要每个 epoch 进行那么多的重采样。最初的原因也可能是 `randperm` 很昂贵。
> 
> However, if you find that sampling in every epoch leads to more stable training, please feel free to send an MR :)但是，如果您发现每个 epoch 中的采样会导致更稳定的训练，请随时发送 MR :)

Thank you very much for your answer！

mahanxin, 2025-02-28:
> While resampling minibatch indices for every epoch is the "right" way, we do not do that since it would marginally improve the performance. The number of mini-batches is small while the size of each [...]

Thank you very much for your answer

swstbecrpgmail, 2025-03-13:
> While resampling minibatch indices for every epoch is the "right" way, we do not do that since it would marginally improve the performance. The number of mini-batches is small while the size of each [...]

Thank you very much for your answer!

closed 2025-03-13 by swstbecrpgmail