leggedrobotics/rsl_rl #33: Actor's Standard Deviation std Not Meeting the Condition of Being ≥ 0 During PPO Training

mitsu3291, 2024-07-11:
I am conducting reinforcement learning for a robot using rsl_rl and isaac lab. While it works fine with simple settings, when I switch to more complex settings (such as Domain Randomization), the following error occurs during training（After some progress in training）, indicating that the actor's standard deviation does not meet the condition of being ≥ 0. Has anyone experienced a similar error?

`Traceback (most recent call last):
  File "/root/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 131, in <module>
    main()
  File "/root/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 123, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/isaac-sim/kit/python/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 153, in learn
    mean_value_loss, mean_surrogate_loss = self.alg.update()
  File "/isaac-sim/kit/python/lib/python3.10/site-packages/rsl_rl/algorithms/ppo.py", line 121, in update
    self.actor_critic.act(obs_batch, masks=masks_batch, hidden_states=hid_states_batch[0])
  File "/isaac-sim/kit/python/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 105, in act
    
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/torch/distributions/normal.py", line 74, in sample
    return torch.normal(self.loc.expand(shape), self.scale.expand(shape))  
RuntimeError: normal expects all elements of std >= 0.0
`

shafeef901, 2024-07-15:
Can confirm I've experienced it too. In my case, I had introduced some sparse rewards to my environment. Not sure that's the cause tho.

felipemohr, 2024-07-21:
Same problem here. When visualizing the training data in tensorboard, I notice that Loss/value_function suddenly goes to infinity

xliu0105, 2024-07-27:
Same problem

AlexanderAbernathy, 2024-07-27:
When facing the error std>=0, check the output 'Value Function Loss' to see whether it's inf or not. If it is inf, there is a solution that you can try. 
Based on the knowledge from issues https://github.com/ray-project/ray/issues/19291 with the fix https://github.com/ray-project/ray/pull/22171 and https://github.com/ray-project/ray/pull/22171/commits/ddd1160b1379125373b4f9c41a375e7ada6a709d, the codes starting at L159 in ppo.py file of rsl_rl (version 2.0.2) need to be modified as follows:

if self.use_clipped_value_loss:
                ### These are the original codes 

                # value_losses = (value_batch - returns_batch).pow(2)
                # value_clipped = target_values_batch + (value_batch - target_values_batch).clamp(-self.clip_param, self.clip_param)
                # value_losses_clipped = (value_clipped - returns_batch).pow(2)
                # value_loss = torch.max(value_losses, value_losses_clipped).mean()

                ### These are the modified codes to the error:  "Value Function Loss: inf"
                ### Details: https://github.com/ray-project/ray/pull/22171
                ###              https://github.com/ray-project/ray/pull/22171/commits/ddd1160b1379125373b4f9c41a375e7ada6a709d
                ###              https://github.com/ray-project/ray/issues/19291
                value_losses = (value_batch - returns_batch).pow(2)
                value_loss = torch.clamp(value_losses, -self.clip_param, self.clip_param).mean()

xliu0105, 2024-07-27:
> When facing the error std>=0, check the output 'Value Function Loss' to see whether it's inf or not. If it is inf, there is a solution that you can try. Based on the knowledge from issues [ray-project/ray#19291](https://github.com/ray-project/ray/issues/19291) with the fix [ray-project/ray#22171](https://github.com/ray-project/ray/pull/22171) and [ray-project/ray@ddd1160](https://github.com/ray-project/ray/commit/ddd1160b1379125373b4f9c41a375e7ada6a709d), the codes starting at L159 in ppo.py file of rsl_rl (version 2.0.2) need to be modified as follows: ![a](https://private-user-images.githubusercontent.com/140694672/352712715-05577235-22c1-4b3f-a0ca-255b98be917a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIwNzIxNTAsIm5iZiI6MTcyMjA3MTg1MCwicGF0aCI6Ii8xNDA2OTQ2NzIvMzUyNzEyNzE1LTA1NTc3MjM1LTIyYzEtNGIzZi1hMGNhLTI1NWI5OGJlOTE3YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyN1QwOTE3MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMDQ3M2ZiNmZhN2QwOWZjMjNmYzI3ZTEzMmExYzYxMjA5MjJmMDA2YjczZDcxZjU2NjliNzlmNmI5NGM5NTAyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.nVDhAn_to-km7QjTcT_tdPAFgVUmLX0zAtzEVQp4H94)

> When facing the error std>=0, check the output 'Value Function Loss' to see whether it's inf or not. If it is inf, there is a solution that you can try. Based on the knowledge from issues [ray-project/ray#19291](https://github.com/ray-project/ray/issues/19291) with the fix [ray-project/ray#22171](https://github.com/ray-project/ray/pull/22171) and [ray-project/ray@ddd1160](https://github.com/ray-project/ray/commit/ddd1160b1379125373b4f9c41a375e7ada6a709d), the codes starting at L159 in ppo.py file of rsl_rl (version 2.0.2) need to be modified as follows: ![a](https://private-user-images.githubusercontent.com/140694672/352712715-05577235-22c1-4b3f-a0ca-255b98be917a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIwNzIxNTAsIm5iZiI6MTcyMjA3MTg1MCwicGF0aCI6Ii8xNDA2OTQ2NzIvMzUyNzEyNzE1LTA1NTc3MjM1LTIyYzEtNGIzZi1hMGNhLTI1NWI5OGJlOTE3YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyN1QwOTE3MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMDQ3M2ZiNmZhN2QwOWZjMjNmYzI3ZTEzMmExYzYxMjA5MjJmMDA2YjczZDcxZjU2NjliNzlmNmI5NGM5NTAyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.nVDhAn_to-km7QjTcT_tdPAFgVUmLX0zAtzEVQp4H94)

Thanks for your answer, I'll try it out

AlexanderAbernathy, 2024-07-27:
When facing the error std>=0, check the output 'Value Function Loss' to see whether it's inf or not. If it is inf, there is a solution that you can try. Based on the knowledge from issues https://github.com/ray-project/ray/issues/19291 with the fix https://github.com/ray-project/ray/pull/22171 and [ray-project/ray@ddd1160](https://github.com/ray-project/ray/commit/ddd1160b1379125373b4f9c41a375e7ada6a709d), the codes starting at L159 in ppo.py file of rsl_rl (version 2.0.2) need to be modified as follows:
[image]

AlexanderAbernathy, 2024-07-29:
> When facing the error std>=0, check the output 'Value Function Loss' to see whether it's inf or not. If it is inf, there is a solution that you can try. Based on the knowledge from issues [ray-project/ray#19291](https://github.com/ray-project/ray/issues/19291) with the fix [ray-project/ray#22171](https://github.com/ray-project/ray/pull/22171) and [ray-project/ray@ddd1160](https://github.com/ray-project/ray/commit/ddd1160b1379125373b4f9c41a375e7ada6a709d), the codes starting at L159 in ppo.py file of rsl_rl (version 2.0.2) need to be modified as follows: ![b](https://private-user-images.githubusercontent.com/140694672/352713352-f7228dfb-2847-4aad-9fe7-80010aefc1ca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIyMjA5ODIsIm5iZiI6MTcyMjIyMDY4MiwicGF0aCI6Ii8xNDA2OTQ2NzIvMzUyNzEzMzUyLWY3MjI4ZGZiLTI4NDctNGFhZC05ZmU3LTgwMDEwYWVmYzFjYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcyOVQwMjM4MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iODdlYjU1NDhjOGVhNTZlMzYwMmRhMWRkYThmZWNhZDQ3M2Q5Nzg1MDcxOThmZmFhZTZjN2JmZjFhNGYwNzE2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2RB2Bdo5FQTCZPPqEq7PDBVVHgeynulNs97lhh8aysg)

Note this kind of method may not work and it may reduce the learning speed. I've tested it using parameters 'iteration : 30000' & 'num_envs : 12000 to 30000' for training my own robot. The training process randomly failed between 1,000 and 18,000 iterations. I've checked the 'value batch' and 'return batch'. Once the training failed, these two values showed very large positive or negative numbers. I ultimately completed the entire training process by modifying the reward and penalty. Since I'm still fresh to the RL, I don't know exactly what happened. By the way, I've tested modifying the hyperparameters of PPO and the structure of networks. It didn't work. I would greatly appreciate it if someone could provide some information on this topic. 

There is an unsuitable method to ensure the training proceeds. When std >= 0 and the Value Function Loss shows inf, you can first adjust some parameters in the project and then use --resume to load the checkpoint and continue training.

weifeng-lt, 2024-08-15:
Adding code actions = torch.clip(actions, min=-6.28, max=6.28) before env.step(actions) seems to help. And it is better to add a penalty to actions to prevent the actor model from outputting too large values.

cyhunblr, 2024-11-28:
+

ClemensSchwarke, 2025-05-15:
We now added logarithmic noise standard deviation, which prevents negative values.

closed 2025-05-15 by ClemensSchwarke

joco163, 2025-05-27:
> We now added logarithmic noise standard deviation, which prevents negative values.

Could someone please give short summary of how and where to implement this? I am facing a similar issue reported here.