leggedrobotics/rsl_rl #38: Huge performance gaps between algorithms and master on base IsaacLab tasks

vassil-atn, 2024-08-30:
Hi,
First of all, thank you for the algorithm implementations! I've tried migrating from the master branch to the algorithms one (since it supports more algorithms than just PPO), and testing it on the flat terrain ANYmal locomotion env in Isaac Lab. For some reason there is a huge gap in performance between the master branch PPO and the algorithms branch PPO - the latter exhibits a much larger reward variance and does not learn to follow the velocity commands at all in the 300 update steps. I use the same environment for both and the LegacyRunner for the algorithms branch. I've uploaded two screenshots of the training curves - interestingly, they start off the same but then diverse after several update steps.

Has anyone encountered this issue before?

[image]
[image]

vassil-atn, 2024-08-30:
I actually found the issue - num_learning_epochs is not used as a parameter in the new version (so each batch is used once only for updating). If you modify the code and run it with 5 epochs (like the master branch PPO), it works fine:
[image]

On that note, is there a particular reason for changing the behaviour to not allow training over multiple epochs?

closed 2024-08-30 by vassil-atn

XanderWangXY, 2024-12-09:
Hello！I have met the same problem. Where are the wrong codes?

vassil-atn, 2024-12-09:
Hi, you can change the self.storage.batch_generator() call to also iterate over a given number of epochs (as an outer loop surrounding the `for i in range(batch_count):`

vassil-atn, 2024-12-10:
Honestly I haven't tried anything other than PPO for a while - my previous experience was that SAC did not perform as well, which might have had to do with buffer sizes - i.e. with 4k environments and 24 steps per env per rollout, a standard 1e6 step buffer is effectively replaced every 10 rollouts.