leggedrobotics/rsl_rl #42: torch.load in the runner will throw warning due to ACE security vuln

KyleM73, 2024-10-13:
torch.load allows arbitrary code execution when the argument weights_only is set to False (default). The default value will switch to True in a future torch release. People wishing to get rid of the error (and eliminate the risk of ACE when loading files you did not train yourself) can change the torch.load call in the runner.load method to include the argument weights_only=True as a default.

Specifically:

```
def load(self, path, load_optimizer=True):
        loaded_dict = torch.load(path)
```

should become 

```
def load(self, path, load_optimizer=True):
        loaded_dict = torch.load(path, weights_only=True)
```

Mayankm96, 2025-02-08:
Hi! Thanks for bringing this to our notice.

Does `wieghts_only` affect the usecase when we want to resume training from a checkpoint?

ClemensSchwarke, 2025-05-15:
Closing due to inactivity.

closed 2025-05-15 by ClemensSchwarke