leggedrobotics/rsl_rl #48: CPU usage

DTSwink, 2025-01-09:
a quick performance monitoring indicated that the bottleneck was concentrated on 1 of my CPU
by profiling the python code i get that the 2 slowest functions are `torch.as_tensor` and `_validate_2D_io_variables()`
which seem to be called at every step, is it intended ? thanks

the log :

```
2131668/1608293    2.661    0.000    3.953    0.000 /home/me/Documents/myenv2/lib/python3.12/site-packages/taichi/lang/kernel_impl.py:1214(_getattr)
     2400    0.024    0.000    5.266    0.002 /home/me/Documents/rsl_rl/rsl_rl/algorithms/ppo.py:90(act)
     4400   10.881    0.002   13.065    0.003 /home/me/Documents/rsl_rl/rsl_rl/modules/actor_critic.py:119(update_distribution)
     4400    0.010    0.000   13.644    0.003 /home/me/Documents/rsl_rl/rsl_rl/modules/actor_critic.py:134(act)
    83974   22.945    0.000   22.945    0.000 {method 'item' of 'torch._C.TensorBase' objects}
    67959   31.006    0.000   31.006    0.000 {built-in method torch.as_tensor}
    13558    0.111    0.000   31.402    0.002 /home/me/Documents/Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py:3205(_validate_2D_io_variables)
```

ClemensSchwarke, 2025-05-15:
Hi @DTSwink,
thanks for compiling this. What GPU are you using for training?

ClemensSchwarke, 2025-05-26:
Closing due to inactivity.

closed 2025-05-26 by ClemensSchwarke