leggedrobotics/rsl_rl #58: Which branch should be used?

AugustDG, 2025-02-11:
I saw that both `main` and `algorithms` are quite different, which one do you guys recommend using?

Mayankm96, 2025-02-11:
We usually use the `main` branch (for its PPO implementation).

The `algorithms` branch was done by one of the students in our group. However, as they aren't part of the lab anymore, noone really is maintaining it anymore. If you need implementations of other algorithms (like SAC, TD3), I suggest using another library such as `skrl` instead.

AugustDG, 2025-02-11:
Oh awesome, thanks @Mayankm96 ! If I ever decide to refactor the `main` branch on my fork (for some more customizability), would you guys be interested in a PR?

Mayankm96, 2025-02-11:
Definitely! We are happy to accept contributions of all sorts -- bug fixes, new features as well as code improvements :)

AugustDG, 2025-02-11:
Awesome! Thanks :)

closed 2025-02-11 by AugustDG