leggedrobotics/rsl_rl #59: Add a config option to clamp actor network output

reeceomahoney, 2025-02-11:
When using the OnPolicyRunner for some envs and reward functions, the agent can often get stuck in a feedback loop where the value of the action explodes causing training to crash. Could be easily fixed by adding a torch.clamp in the update_distribution function of ActorCritic, along with a config option to set whether this is active or not. Would be happy to submit a pull request if you think this is a good idea.

reeceomahoney, 2025-02-13:
I am assuming you mean whether to add it to the OnPolicyRunner's env interaction or the actor critic directly? I think it might be easier to add it in the actor critic directly where the model is called as all other calls are downstream of this and then we wouldn't have to worry about duplicating this change for all of them?

Let me know if that sounds reasonable.

ClemensSchwarke, 2025-05-26:
Closing due to inactivity.

closed 2025-05-26 by ClemensSchwarke