leggedrobotics/rsl_rl #89: Unexpected mean reward behavior after resuming training

lorenzo-bianchi, 2025-05-07:
Hello everyone,
I'm using IsaacSim 4.5.0 and RSL-RL 2.3.1 to train the Ant model, apply some modifications to it, and then fine-tune the policy.

In my first test, I interrupted the training and resumed it from the last checkpoint.
What I don't understand is why, after restarting, the mean reward initially drops to (and also below) zero. It then takes around 27 steps to reach ~15k, and several hundred more steps to return to the reward level achieved before the interruption.

Could you please explain why this behavior happens?
Thanks for your answers!

[image]

[image]

Mayankm96, 2025-05-07:
Most likely because of the parameter `init_at_random_ep_len` here which is set to True: https://github.com/leggedrobotics/rsl_rl/blob/main/rsl_rl/runners/on_policy_runner.py#L132

lorenzo-bianchi, 2025-05-07:
Hi Mayank,
Thank you for the quick reply.
I've tried to set that parameter to False, but the result is even worse than before.

[image]

lorenzo-bianchi, 2025-05-07:
It's the original `Isaac-Ant-Direct-v0` without any customizations, curricula, or empirical normalization.

Mayankm96, 2025-05-09:
We'd need to investigate where this is originating from. Most likely some "global" count for optimizer is not being saved/loaded correctly. I wouldn't have the bandwidth to take a look until later this month...

@jsmith-bdai if I recall, you also saw similar issues with your team, right?

ClemensSchwarke, 2025-05-16:
Hi @lorenzo-bianchi, I just trained the `Isaac-Ant-Direct-v0` environment for 1000 iterations and then continued for another 1000 on the latest IsaacLab and rsl_rl releases. Everything works fine for me. An initial drop in reward is expected and not related to the actual policy performance, but rather due to how logging works. Are you sure that you are loading the correct checkpoint?

The reward I am getting for reference:

[image]

lorenzo-bianchi, 2025-05-16:
Hi @ClemensSchwarke, thanks for your reply. I think it was the correct checkpoint because I didn't specify the initial step manually for the second training, and it started correctly at 1500. The initial drop is not a problem, but in my case the previous reward is reached only after many steps. I'll try to investigate more.

ClemensSchwarke, 2025-05-19:
Let me know in case you need help or have further details that would help me reproduce the behavior you are seeing.

Hellod035, 2025-06-08:
same error. The resume training does not seem to be perfect. In extreme cases, the model seems to need to learn from scratch again.

ozhanozen, 2025-06-17:
I have the same issue, the learning is deteriorated significantly upon resume. Having custom environment with empirical normalization.

fgmn, 2025-09-02:
Could this be because I used curriculum settings in my custom environment but didn’t load the previous curriculum state when resuming training?

ClemensSchwarke, 2025-10-15:
As this behavior does not occur on Isaac-Ant-Direct-v0 with and without normalization I would close the issue for now.