leggedrobotics/rsl_rl #92: Combine CNN-based camera input with ActorCriticRecurrent

PanagiotisKotsaris, 2025-05-14:
I’m currently using the CNN implementation from [ #74 ] to preprocess depth-camera frames. Previously, I was feeding all observations (including images) into the ActorCriticRecurrent module.

I’d like to feed the raw camera observation into the CNN encoder, then concatenate its output embedding with my other (non-image) observations, and finally pass that combined vector through the rest of the ActorCriticRecurrent network. Is this straightforward?

ClemensSchwarke, 2025-05-15:
Hi Panagiotis, yes that should be pretty straightforward. We are also currently working on a release that would make this even easier :)

ClemensSchwarke, 2025-05-15:
Sounds good!

closed 2025-05-15 by ClemensSchwarke