leggedrobotics/rsl_rl #93: Error while using empirical_normalization with IsaacLab

shafeef901, 2025-05-16:
**Error**
I have been using IsaacLab with RSL-RL for quite some time now. I used `empiricial_normalization=True` in one of my envs.
After the latest upgrade of rsl_rl version with the distillation, it introduces a new error:

```
KeyError: 'privileged_obs_norm_state_dict'
```

**Solution**
The error is straightforward since i didn't use any privileged observations for my previous env. To enable backward compatability, this error can be resolved using simple checks if the `loaded_dict` and `saved_dict` in `OnPolicyRunner` has the aforementioned key that causes the error such as below:

```python
if self.empirical_normalization:
    if resumed_training:
        # if a previous training is resumed, the actor/student normalizer is loaded for the actor/student
        # and the critic/teacher normalizer is loaded for the critic/teacher
        self.obs_normalizer.load_state_dict(loaded_dict["obs_norm_state_dict"])
        if loaded_dict.get("privileged_obs_norm_state_dict") is not None:
            self.privileged_obs_normalizer.load_state_dict(loaded_dict["privileged_obs_norm_state_dict"])
```

ClemensSchwarke, 2025-05-16:
Hi, I am not sure if I fully understand the issue. Why would your training not have the normalizer for priviledged observations?

shafeef901, 2025-05-16:
So if I am not wrong, the normaliser for privileged observations was introduced recently, and the `privileged_obs_norm_state_dict` only exists for newly trained models. While playing one of my old models, I had the error pop up. But I guess, it shouldn't really concern newly trained models with the latest version of rsl_rl.

ClemensSchwarke, 2025-05-16:
Ah I see, thanks. The normalization dictionary was renamed to work with PPO and Distillation at the same time. Your old model should have a dictionary called something like `critic_obs_norm_dict`. We try to backwards compatible in terms of training scripts/configurations etc., also ensuring compatibility with old models would complicate the code base quite a lot in my opinion.

closed 2025-05-19 by ClemensSchwarke