leggedrobotics/rsl_rl #103: Support of multi-obs input

YifeiNie, 2025-07-28:
Hello! In the newest version v3.0.0, rsl-rl uses TensorDict to manage observation, the release page says "Different observation groups with different shapes can thus be handled seemlessly, e.g., vision inputs", which means rsl-rl can handle different shape of observations stored in a dictionary. 
```
    @abstractmethod
    def get_observations(self) -> TensorDict:
```
However, the code seems do not have this support, most modules have codes as follows :
```
    for obs_group in obs_groups["policy"]:
        assert len(obs[obs_group].shape) == 2, "The ActorCritic module only supports 1D observations."
        num_actor_obs += obs[obs_group].shape[-1]
    num_critic_obs = 0
    for obs_group in obs_groups["critic"]:
        assert len(obs[obs_group].shape) == 2, "The ActorCritic module only supports 1D observations."
        num_critic_obs += obs[obs_group].shape[-1]
```
I believe it will bring great advantages if the multi-obs input is supported, looking forward to the reply!
Thanks!

Mayankm96, 2025-07-30:
It is quite hard to keep the flexibility for NN designing (i.e. how the different obs groups get propogated). To not be prescriptive in any form, you need to make a duplicate class for every case right now. If you have better solutions/proposals, we would be interested to discuss.

An example of above:

```python
# Copyright (c) 2021-2025, ETH Zurich and NVIDIA CORPORATION
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

from __future__ import annotations

import os
import torch
import torch.nn as nn
import warnings
from torch.distributions import Normal

from rsl_rl.networks import CNN, MLP, EmpiricalNormalization, Memory
from rsl_rl.utils import unpad_trajectories

class ActorCriticPerceptiveRecurrent(nn.Module):
    is_recurrent = True

    def __init__(
        self,
        obs,
        obs_groups,
        num_actions,
        actor_obs_normalization=False,
        critic_obs_normalization=False,
        actor_hidden_dims=[256, 256, 256],
        critic_hidden_dims=[256, 256, 256],
        activation="elu",
        init_noise_std=1.0,
        noise_std_type: str = "scalar",
        rnn_type="lstm",
        rnn_hidden_dim=256,
        rnn_num_layers=1,
        num_channels_per_image=1,
        **kwargs,
    ):
        if "rnn_hidden_size" in kwargs:
            warnings.warn(
                "The argument `rnn_hidden_size` is deprecated and will be removed in a future version. "
                "Please use `rnn_hidden_dim` instead.",
                DeprecationWarning,
            )
            if rnn_hidden_dim == 256:  # Only override if the new argument is at its default
                rnn_hidden_dim = kwargs.pop("rnn_hidden_size")
        if kwargs:
            print(
                "ActorCriticPerceptiveRecurrent.__init__ got unexpected arguments, which will be ignored: "
                + str(kwargs.keys()),
            )
        super().__init__()

        # get the observation dimensions
        self.obs_groups = obs_groups
        self.num_channels_per_image = num_channels_per_image

        if set(obs_groups["policy"]) != {"policy", "image"}:
            raise ValueError(
                "The StudentTeacher module only supports 'policy' and 'image' observations. Received:"
                f" {obs_groups['policy']}"
            )

        # calculate proprioception
        assert len(obs["policy"].shape) == 2, "The StudentTeacher module only supports 1D observations."
        num_actor_prop_obs = obs["policy"].shape[-1]
[... 171 lines cut ...]
        if self.critic_obs_normalization:
            critic_obs = self.get_critic_obs(obs)
            self.critic_obs_normalizer.update(critic_obs)

    ##
    # Loading and exporting model.
    ##

    def load_state_dict(self, state_dict, strict=True):
        """Load the parameters of the actor-critic model.

        Args:
            state_dict (dict): State dictionary of the model.
            strict (bool): Whether to strictly enforce that the keys in state_dict match the keys returned by this
                           module's state_dict() function.

        Returns:
            bool: Whether this training resumes a previous training. This flag is used by the `load()` function of
                  `OnPolicyRunner` to determine how to load further parameters (relevant for, e.g., distillation).
        """
        super().load_state_dict(state_dict, strict=strict)
        return True
```

More integrated examples with IsaacLab will follow in the coming month.

YifeiNie, 2025-08-11:
@Mayankm96 Thanks for the inference provided and the reply, it greatly helped and the old problems were solved
However for rnn based networks, just realize a custom network will not work since updating hidden states and truncating and aligning sequences are involved.
The existing recurrent actor critic seems has some problems in my projects, I'm now working on it
All in all, thanks for your reply and looking forward to the new version

raidenLight, 2025-08-12:
Hello!
In the example code you provided, it seems that the image is first encoded by the encoder, proprioceptive inputs are processed through the RNN network, then the two outputs are concatenated and passed through empirical normalization, and finally output by an MLP-based actor. However, I’m concerned this might affect training performance. Theoretically, the original observation information should be normalized before being fed into the entire network, and only then split for further processing. Also, the `get_actor_obs` and `get_critic_obs` functions are rather misleading.

Here are my suggestions:

1. Could we directly apply empirical normalization to the **original observation TensorDict**? This should be quite straightforward — we just need to flatten the entire TensorDict, normalize it, and then restore it to its original shape:

```python
class TensorDictGlobalNorm(nn.Module):
    """
    Apply EmpiricalNormalization to the whole TensorDict as if it's a single vector.
    This will flatten all tensors inside the TensorDict, concatenate, normalize, then reshape back.
    """

    def __init__(self, example_tensordict: TensorDict, eps=1e-2, until=None):
        super().__init__()
        # Store original shape for each key
        self.shapes = {k: v.shape[1:] for k, v in example_tensordict.items()}
        total_dim = sum(v[0].numel() if isinstance(v, tuple) else torch.tensor(v).numel()
                        for v in self.shapes.values())
        self.norm = EmpiricalNormalization(shape=(total_dim,), eps=eps, until=until)

    def _flatten_td(self, td: TensorDict) -> torch.Tensor:
        # Flatten and concatenate in key order
        flat_tensors = [td[k].reshape(td.batch_size[0], -1) for k in self.shapes.keys()]
        return torch.cat(flat_tensors, dim=1)

    def _unflatten_tensor(self, flat: torch.Tensor) -> dict:
        out = {}
        idx = 0
        for k, shape in self.shapes.items():
            size = torch.tensor(shape).prod().item()
            out[k] = flat[:, idx: idx + size].reshape(-1, *shape)
            idx += size
        return out

    def forward(self, td: TensorDict) -> TensorDict:
        td = td.clone()
        flat = self._flatten_td(td)
        flat_normed = self.norm(flat)
        for k, v in self._unflatten_tensor(flat_normed).items():
            td.set(k, v)
        return td

    def update(self, td: TensorDict):
        flat = self._flatten_td(td)
        self.norm.update(flat)

    def inverse(self, td: TensorDict) -> TensorDict:
        td = td.clone()
        flat = self._flatten_td(td)
        flat_inv = self.norm.inverse(flat)
        for k, v in self._unflatten_tensor(flat_inv).items():
            td.set(k, v)
        return td
```

Alternatively, we could maintain a normalization class **per key** in the observation dict, which may be more efficient:

```python
class TensorDictEmpiricalNorm(nn.Module):
    """Apply EmpiricalNormalization to a specific key in TensorDict."""

    def __init__(self, key, shape, eps=1e-2, until=None):
        super().__init__()
        self.key = key
        self.norm = EmpiricalNormalization(shape, eps=eps, until=until)

    def forward(self, tensordict: TensorDict):
        tensordict = tensordict.clone()
[... 133 lines cut ...]
        if self.noise_std_type == "scalar":
            std = self.std.expand_as(mean)
        else:
            std = torch.exp(self.log_std).expand_as(mean)
        self.distribution = Normal(mean, std)

    def act(self, obs, masks=None, hidden_states=None):
        mean = self.actor_net(obs, masks, hidden_states)
        self.update_distribution(mean)
        return self.distribution.sample()

    def act_inference(self, obs):
        return self.actor_net(obs)

    def evaluate(self, obs):
        return self.critic_net(obs)

    def update_normalization(self, obs):
        if isinstance(self.actor_net.obs_normalizer, TensorDictGlobalNorm):
            self.actor_net.obs_normalizer.update(obs)
        if isinstance(self.critic_net.obs_normalizer, TensorDictGlobalNorm):
            self.critic_net.obs_normalizer.update(obs)

```
What do you think of this?