leggedrobotics/rsl_rl #113: How can I use the Recurrent_mini_batch_generator

pumfish, 2025-09-14:
Hello, thanks your work.
I'm working on your code to deploy an RNN-style PPO. I noticed there has an ```Recurrent_mini_batch_generator``` in the RolloutStorage, and I also noticed that the PPO ```update``` function already includes the corresponding checks in lines 201-205.
But I found that in the ```update``` function, many operation is for "batch-first", but the output of ```Recurrent_mini_batch_generator``` is "time-first". Am I misunderstanding something, or does rsl_rl currently not provide an algorithm class that supports RNN-PPO?
https://github.com/leggedrobotics/rsl_rl/blob/d38a378b13a60ea570923754184f84f48979ad74/rsl_rl/algorithms/ppo.py#L186-L427

pumfish, 2025-10-15:
Thanks for your reply.

I tried to use ```ActorCriticRecurrent``` to implement an RNN-PPO, but I found that the training success rate (SR) always remains 0, whereas standard PPO can successfully train on the same task.

I also noticed a similar issue mentioned here: https://github.com/leggedrobotics/rsl_rl/issues/121

I am trying to understand why RNN-PPO performs poorly. I am not sure whether this is due to hyperparameter settings or a potential bug in the code. However, I don’t think that using the same hyperparameters as standard PPO should cause RNN-PPO to completely fail while PPO trains successfully.

Any suggestions on why RNN-PPO fails while PPO works would be greatly appreciated.

ClemensSchwarke, 2025-10-15:
Hi @pumfish,
Yes, as mentioned in the other issue, using symmetry is currently broken for RNN training unfortunately. However, when not using symmetry everything should work out of the box. It is hard to tell the cause of your problem from the provided info, but what I would recommend is to train a simple default task (e.g. cartpole) to validate that the library is doing what it is supposed to do.